Alarm-Siren/6502-kicad-library #12: Symbol library '6502' failed to load

someyob, 2023-07-14:
Following the instructions here, using KiCAD 6, get dialog box "Expecting '(' in 'C:\....\65xx.kicad_sym', line 7, offset 1."

Alarm-Siren, 2023-07-15:
Hi,

I've been unable to replicate this problem, it seems to work just fine in my own copy of KiCAD 6. Which specific version are you using? e.g. 6.0.8, 6.0.11, etc?

someyob, 2023-07-15:
6.0.11
I first installed version 7 and had the same error, so I went with 6 to see if  it was a compatibility issue.
I'm new to KiCAD, so it may be that I'm doing something wrong.

Alarm-Siren, 2023-07-15:
As it happens, I've just uploaded a new version of the library anyway, which has a different installation procedure (it now uses kicad's integrated package manager), so give that a go. I've just tested the new package in Kicads 6.0.8 and 7.0.2 and it works for me following the new instructions given.

https://github.com/Alarm-Siren/6502-kicad-library/releases/tag/v3.0.0

someyob, 2023-07-15:
Hi, appreciate the response, thanks very much.  I won't be able to dig into this for a few days, but will report back when I do.

someyob, 2023-08-01:
Hi, ran the library install under KiCAD 7.0.6, and all appears to be well.  Thanks!

closed 2023-08-01 by Alarm-Siren