Alarm-Siren/6502-kicad-library #13: Archive does not contain a valid metadata.json file

PopovMP, 2024-02-17:
I cannot import the library.

It gives "Archive does not contain a valid metadata.json file"

Windows 11. KiCad v7.0

Alarm-Siren, 2024-02-18:
Hi @PopovMP, I'm sorry to hear that you're having trouble.

Can I ask you to confirm that you've followed the installation instructions in the [readme file](https://github.com/Alarm-Siren/6502-kicad-library?tab=readme-ov-file#kicad-7) exactly? I have been unable to replicate the issue using KiCad 7.0.10 when following those instructions.

Note that you *have* to download the correct release package as described in the instructions. If you download the "source code" zip file that will, indeed, give you the error you experienced.

PopovMP, 2024-02-21:
Thank you for your response.
It worked after restarting KiCad.