Alarm-Siren/6502-kicad-library #14: 6502 active low on pin 38 and missing pin 21

tomse, 2024-03-15:
Pin 38 is active low in your part, but it's actually active high

[image]

Pin 21 is missing (VSS)
[image]

Alarm-Siren, 2024-03-16:
Hi @tomse, thanks for getting in touch. I have reviewed your comments, and I believe in both cases the library is correct.

With respect to pin 38, you said that the pin should be marked as active high. However, the [datasheet](http://www.6502.org/documents/datasheets/mos/mos_6500_mpu_mar_1980.pdf) says otherwise on page 6...

> **Set Overflow Flag (S.O.)**
> A NEGATIVE going edge on this input sets the overflow bit in the Status Code Register. This signal is sampled on the trailing
edge of [phi0]

I appreciate that the signal name is not given with an overbar in the datasheet, but the description clearly makes it an active low signal so I have labelled it as such on the symbol.

With respect to pin 21, I can assure you the pin is in fact present. It uses an invisible stacked pin as per the [Kicad Library Conventions section 4.3](https://klc.kicad.org/symbol/s4/s4.3/). You just can't see it in the normal symbol view, but if you open the symbol in the symbol editor it can be seen lurking underneath pin 1. Because of this arrangement, connecting a net to pin 1 will also connect to pin 21.

[image]

tomse, 2024-03-17:
ok.
pin 21 isn't visible on the diagram, if you use the schematic for troubleshooting, you'd need to lookup the component unless you know the pinout by heart.

most diagrams I've seen and component pinouts that use the 6502, show pin 38 as non negated (even the drawing on page 9)
I have no other arguments for/against this other than "majority does this" :-) 

Thanks for making the library though :-)

Alarm-Siren, 2024-03-17:
> pin 21 isn't visible on the diagram, if you use the schematic for troubleshooting, you'd need to lookup the component unless you know the pinout by heart.

Nevertheless, this is correct symbol design going by the Kicad Library Conventions - which the entire Kicad standard library also uses.

> most diagrams I've seen and component pinouts that use the 6502, show pin 38 as non negated (even the drawing on page 9)
I have no other arguments for/against this other than "majority does this" :-)

I do not strictly adhere to the datasheet names, or even "majority use", when compiling the symbols, prefering to use what I judge to be the most correct and/or non-misleading signal names.

For evidence of the SO pin being named the other way please see the datasheet for the [W65C02S](http://www.westerndesigncenter.com/wdc/documentation/w65c02s.pdf) (which is basically the modern CMOS version of the same chip) which, on pages 12-14 shows the SO pin as "SOB" (which stands for "Set Overflow Bar", where Bar means Overbar, i.e. negative - all the other negative pins are similarly suffixed with 'B' in this datasheet).

The 6502 ecosystem is notorious for having misleading and confusing naming conventions, but I don't see that as a reason to perpetuate the situation.

Of course in both cases its easy enough for you to modify your local copy of the library if you prefer.

> Thanks for making the library though :-)

You're most welcome and, despite the aforementioned niggles, I hope you've found it useful 👍

closed 2024-03-17 by Alarm-Siren