Alarm-Siren/6502-kicad-library #15: Kicad 8 install instructions on a Mac

sharpdad, 2024-07-04:
Thank you very much for this library! 

I am a new Kicad 8 user on a Mac and it took me a while to figure out how to install it.

There is no Preferences->Preferences on a Mac for step 2 so I could not figure out how to do steps 2-7.

I did successfully install the library using the main panel with the "Plugin and Content Manager" at the bottom of the screen. Macs automatically unzip zip files when downloading so I needed to re-zip and/or re-download the file but it all seems to be working after using the zip. 

[image]

sharpdad, 2024-07-04:
Now that I wrote this, on the Mac you need to do Settings which brings up the Preferences Panel and then choose the "Packages and Updates" option which has the options from steps 4-6.

[image]

Alarm-Siren, 2024-07-16:
Hi @sharpdad 

Thanks for submitting this info.

Yeah, I need to update the instructions. I had a similar issue report on my other Kicad repo, too. Unfortunately I don't have a Mac, nor know anyone who does, so I'll have to rely on other people to tell me if I get the instructions right!